CodyTolene/Flipper-Zero-Camera-Suite #24: Store Greyscale or Color images

leedave, 2023-08-27:
Not sure how well this is possible, from my knowledge the ESP32 Cam should not only support GRAYSCALE , but also RGB888 (images with 3 color channels of 0 - 255) or even direct a stream as JPEG. 

Would be cool to be able select if the image stored by the app is in color or maybe even a color JPEG

CodyTolene, 2023-09-27:
This would be a nice feature! Could add a selectable save type option for it (save as displayed on Flipper, or an image as best possible by the esp32 cam -> jpeg?)

I wonder if it would also be possible to not only stream the image, but broadcast it on a certain frequency. Flipper Zoom?

CodyTolene, 2023-09-28:
I'm not sure if it's possible to save the JPEG on the Flipper via the I/O as it might be impactful on the performance to stream the full channel data. However it should be less impactful and a better approach if the camera has a memory card to save jpeg directly from the esp32 to it. I will look into this as soon as I can.

CodyTolene, 2023-10-03:
Staged new functionality for this here:
https://github.com/CodyTolene/Flipper-Zero-Camera-Suite/pull/25/files#diff-57c15f919300b26a2bdce1d377a017ea8f8769815e8e4c108925f6d1cfbea243

It's not implemented yet, but most of the functionality is there it just needs to be tested. Will be able to look into this more as time permits. Thanks for the good suggestion.