CodyTolene/Flipper-Zero-Camera-Suite #27: Mirrored Image

dezign999, 2023-10-28:
I have noticed that the camera will randomly display a mirrored image. Although it's hard to notice, it was brought to my attention when trying to capture the an image with text. Exiting and restarting the app will sometimes resolve this, however the frequency at which it happens outweighs the time that it's in the correct orientation.

CodyTolene, 2023-11-02:
Thank you for opening this ticket I will try to look into this soon!

dezign999, 2023-11-03:
No problem. Just an FYI, but the bug does appear in the other camera as well. I’ve noticed that switching between the two can temporarily correct the mirroring issue.

CodyTolene, 2023-11-04:
In my testing it seems that running the older `[ESP32] Camera` app prior to using this new `[ESP32] Camera Suite` is causing some conflicts on the firmware. 

When running `[ESP32] Camera Suite` on a fresh boot without opening `[ESP32] Camera` the issue doesn't occur. I may have to work in some backwards support here. Stay tuned.

dezign999, 2023-11-04:
Wow, that was a nice find. I did launch the older camera app once come to think of it. I remember that it turned on the camera's light even though I have it disabled in your app. Anyway, I did a quick reboot of my flipper and the mirroring was still persistent. Tried pressing and holding the reset button on the camera but still no change. Finally powered down the Flipper and the image is now back to normal. I've put the device to sleep for some time and come back to it 3 times now and each time the camera was the correct orientation. I'm not afraid to call this "User Error," I'll delete the other fap to be safe 😆

CodyTolene, 2023-11-17:
I updated the entire app stemming from your report here, to stabilize things. In my testing of the changes using `[ESP32] Camera` prior to `[ESP32] Camera Suite` you will still get issues, but you can fix them if you open `[ESP32] Camera Suite` and back in and out of the "Open Camera" scene a few times to refresh everything.

For now, I recommend not using the old `[ESP32] Camera` as it seems to be causing issues with new versions of this app and the firmware.

Thanks again for your report and I hope this helps! Cheers

PR #28 (2023-11-17): [27] Bugfix - Horizontal flip when switching between camera apps.
Closes #27 

Changes

- [x] Improve FAP.
- [x] Improve Firmware.
- [x] Improve Firmware flashing utility.
- [x] Improve GitHub actions.
- [x] Look to mitigate issue #27.