CodyTolene/Flipper-Zero-Camera-Suite #30: App not building on toolchain 46.0 f7

doomwastaken, 2023-11-16:
Greetings! I am from Flipper Devices team and would like to say thank you for submitting your app to the catalog. We are finalizing API prior to 1.0 release and currently your application cannot be built for latest toolchain. Currently we are testing 0.95.0-rc and app is not being built for that version. 

We are currently working on automating the messages from our build server, so this one is manual :)

If you have any questions, just tag me

CodyTolene, 2023-11-16:
@doomwastaken I appreciate your report! I've submitted a PR to fix that up here: https://github.com/flipperdevices/flipper-application-catalog/pull/185

Thank you, Cody

CodyTolene, 2023-11-17:
Resolved here: https://github.com/flipperdevices/flipper-application-catalog/pull/187

Thanks again doomwastaken

closed 2023-11-17 by CodyTolene