CodyTolene/Flipper-Zero-Camera-Suite #35: [Feature] WiFi Streaming/Connection

CodyTolene, 2024-01-28:


spitfire, 2024-01-31:
I've tried building firmware from this branch (on commit b348254) and it is failing for me with:

```
In file included from /Users/mieszko/GitHub/Flipper-Zero-Camera-Suite/firmware/firmware.h:10,
                 from /Users/mieszko/GitHub/Flipper-Zero-Camera-Suite/firmware/firmware.ino:1:
/Users/mieszko/GitHub/Flipper-Zero-Camera-Suite/firmware/stream_to_wifi.h:8:10: fatal error: ESPAsyncWebServer.h: No such file or directory
 #include <ESPAsyncWebServer.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: ESPAsyncWebServer.h: No such file or directory
```

Is it a dependency that's missing from the project files?

CodyTolene, 2024-03-17:
Hi @spitfire sorry for the late reply yep those were new files I believe a part of expressif's library. I'm still looking into this feature but will follow up when/if I complete it. Thanks for checking it out!

PR #36 (2024-03-12): [#35] Wi-Fi Features
Integrate Wi-Fi streaming (or attempt to).

Closes #35 

Goals:
- Broadcast an AP to be able to connect to it with Wi-Fi on another device. 
- When connected give the user a webpage to control the camera and view the feed.

Notes:
- Uses new UART available in the Flipper Firmware v0.98-rc+ version. Thanks for your help @Willy-JL!