CodyTolene/Flipper-Zero-Camera-Suite #37: compilation fails with esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory

spitfire, 2024-01-29:
I've tried to run compilation with your `firmware-flash.bat` script. I have no experience with arduino itself (only used it via ESPHome), how do I install missing stuff?

```
C:\Users\mieszko\GitHub\Flipper-Zero-Camera-Suite>firmware-flash.bat
┏┓   ┓    ┏┳┓  ┓
┃ ┏┓┏┫┓┏   ┃ ┏┓┃┏┓┏┓┏┓
┗┛┗┛┗┻┗┫   ┻ ┗┛┗┗ ┛┗┗
       ┛  https://github.com/CodyTolene

Flipper Zero - ESP32-CAM Firmware Flasher - Windows 10+
https://github.com/CodyTolene/Flipper-Zero-Camera-Suite

------------------------------------------------------------------------------
Before you begin please make sure your Flipper Zero is plugged into your PC.
Then on your Flipper Zero, open the GPIO menu and select USB-UART Bridge. In
the USB-UART Bridge config menu, make sure the following configuration is set:
- USB Channel = 1 (on newer firmware)
- Baudrate = Host
- UART Pins = 13,14
- RTS/DTR Pins = None
------------------------------------------------------------------------------

Press any key to continue . . .

Initializing...
Checking configs...
Fetching assets...
Assets already installed. Skipping...
Install to default AI-Thinker ESP32-CAM board with FQBN 'esp32:esp32:esp32cam'? (Y/N): y

Compiling firmware, this will take a moment...

In file included from c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\string:41,
                 from c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:\Users\mieszko\AppData\Local\Temp\arduino-cli\data\packages\esp32\hardware\esp32\2.0.14\cores\esp32/HardwareSerial.h:49,
                 from C:\Users\mieszko\AppData\Local\Temp\arduino-cli\data\packages\esp32\hardware\esp32\2.0.14\cores\esp32/Arduino.h:184,
                 from C:\Users\mieszko\AppData\Local\Temp\arduino\sketches\26F6772AB900998463D0FE4FC6F8B8E4\sketch\firmware.ino.cpp:1:
c:\users\mieszko\appdata\local\temp\arduino-cli\data\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory
 #include <bits/c++allocator.h> // Define the base class to std::allocator.
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

CodyTolene, 2024-01-30:
Looks like something went wonky.

Delete this folder:
`c:\users\mieszko\appdata\local\temp\arduino-cli`
and try re-running the script to see if it resolves it.

This is the temp (`%temp%\arduino-cli`) folder that assets for the ESP32-CAM install to. 

Did this happen using the `main` branch (release version) of the code?

spitfire, 2024-01-31:
Tried to delete this folder but it was not present - tried compiling again, and it looks like after it fails your offers to clean up temporary files `Would you like to delete the temporary files? (Y/N): y`. I've modified the `arduino-cli.yaml` as you suggested.

Eventually I've noticed I've been using a different board (esp32-wrover with a ov2640 camera) (sorry about not mentioning that). I've tried setting the board to `esp32:esp32:esp32wrover`, but it still would fail the same way.

I've ordered actual esp32-cam boards, and they should arrive today

Finally I've tried compiling the code in Arduino IDE (with the right board id) and it successfully does and uploads it to my wroom board, but it doesn't seem to work with your app.

I'll test it again when the actually compatible board arrives.

Sorry for the confusion and thanks for making this!

spitfire, 2024-01-31:
Compilation still fails (with the same error) with recommended board, works fine in Arduino IDE.

CodyTolene, 2024-03-17:
Dang I haven't tested it on other boards, I did put in support for it though in hopes that it would work. I super appreciate your testing and feedback on it. Curious of what the differences are in programming.

Also very weird that the batch file didn't work for you, sorry about that, I know it works for most using Windows 10+. I'm really glad the arduino ide worked for you though that's probably got a bit more support for various systems.

Everything was just recently updated to work with the new F0 UART as well so be sure to pick up the latest updates. Thanks again

CodyTolene, 2024-03-17:
Keeping this open, didn't mean to close it.

CodyTolene, 2024-03-17:
Previous title: "compilation fails with esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory"

Simplified a bit but tracked.

spitfire, 2024-03-17:
> Dang I haven't tested it on other boards, I did put in support for it though in hopes that it would work. I super appreciate your testing and feedback on it. Curious of what the differences are in programming.
> 
> 
> 
> Also very weird that the batch file didn't work for you, sorry about that, I know it works for most using Windows 10+. I'm really glad the arduino ide worked for you though that's probably got a bit more support for various systems.
> 
> 
> 
> Everything was just recently updated to work with the new F0 UART as well so be sure to pick up the latest updates. Let me know if that works for you. Thanks again

Hey,just like I said it also failed with the recommended board (got two of these). Have there been any changes related to this that I could test? I was running it on Windows 11 (although I'd love to have it working on my Mac too;)).