CodyTolene/Flipper-Zero-Camera-Suite #54: How to browse bmp files on Flipper.

4k3or3et, 2024-03-15:
Hi All,

I am noob in all sorts of electronics. I just managed to flash ESP32-CAM module with proper firmware after quadrillion tries. It works just fine. Ultra cool that thing is. How do you guys browse those tiny .bmp files on Flipper Zero? I know I can download them to phone or computer but it would be really cool to display them on Flipper display. Any help much appreciated.

CodyTolene, 2024-03-17:
Thank you! You can find the images in the "DCIM" folder at the root of your SD card. You can do that directly or by using the qFlipper app: https://github.com/flipperdevices/qFlipper

Hope that helps

closed 2024-03-17 by CodyTolene

4k3or3et, 2024-03-18:
> Thank you! You can find the images in the "DCIM" folder at the root of your SD card. You can do that directly or by using the qFlipper app: https://github.com/flipperdevices/qFlipper
> 
> Hope that helps

Thank you for your reply. I can find the images but there is no possibility to view then directly on Flipper. This is why I submitted request of new feature:

https://github.com/CodyTolene/Flipper-Zero-Camera-Suite/issues/55

Thank you.